rapp-project/rapp-platform #182: CI - Must fail when tests fail

etsardou, 2015-12-01:
- [ ] Unit & functional - check ROS
- [ ] Integration - when something fails, return error code

athallas, 2015-12-03:
>Policy CMP0046 is not set: Error on non-existent dependency in add_dependencies

This is a catkin [issue](https://github.com/ros/catkin/issues/761)

athallas, 2015-12-03:
>The directory '/home/travis/.cache/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.

Not supported in the current version. [link](https://docs.travis-ci.com/user/caching/)

etsardou, 2015-12-03:
Closing this. Made the remaining two new issues.

closed 2015-12-03 by etsardou