rapp-project/rapp-platform #185: RAPP API Python documentation generation

etsardou, 2015-12-03:
Add to documentation script and update the webpage accordingly. The folder is: https://github.com/rapp-project/rapp-api/tree/master/python

Do this for the C++ code as well (I think @alexge233 has documented the code in Doxygen style). 

Something similar will happen for js. @lpenteri does the JS code have documentation that can be extracted in any form (e.g. Doxygen or JSDoc)?

alexge233, 2015-12-03:
I've documented most (if not all) of the C++ API, using JavaDoc style (What Doxygen uses) comments.
Are we going to set-up a dedicated Doxygen website?
@lpenteri Should have the Documentation ready soon, but we also need an install script for the JS API (NodeJS, and some modules).

etsardou, 2015-12-04:
We have already setup a RAPP Platform webpage located [here](http://rapp-project.github.io/rapp-platform/). The documentation that exists there is automatically updated each time we push at the ```master``` branch, via [travis](https://travis-ci.org/rapp-project/rapp-platform/branches).

I think the RAPP API (Doxygen generated) should be in [this page](http://rapp-project.github.io/rapp-platform/documentation.html), as the API goes along with the Platform. Also you can have a dedicated section for tutorials etc. in the rapp-api wiki (I think you have started that).

Since you already have Doxygen style documents we will try to create the docs and integrate them in Travis and the webpage. Regarding the JS I think we need a script to generate the docs (something like [this](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/Makefile) ), in order to call it wherever appropriate.

From what I see we are going to make the release next week, thus the more you can do till then, the merrier. Of course, since the documentation will be automatically updated hereinafter you can update it in Github and the changes will be pushed when we push to master.

Regarding the rapp-api CI you should create a travis.yml file like [this](https://github.com/rapp-project/rapp-platform/blob/master/.travis.yml) if you want the CI to build your code and run the tests. When you create it I will help you to deploy it.

lpenteri, 2015-12-05:
There are comments almost everywhere in the JS API code already using JavaDoc style as well, so Doxygen should be able to extract the documentation. I will test that this is indeed working these days and check if anything is missing and I am also working on a makefile like the one in the link you sent for the platform web services. Finally regarding the install script that Alex mentioned is already up in the root folder of the JS API in ortelio branch, but I will also include it in the makefile to be executed with "make deps".

klpanagi, 2015-12-05:
Completed task for Python Rapp API: rapp-project/rapp-api#5

closed 2015-12-07 by athallas