rapp-project/rapp-platform #192: Scripts' individual READMEs are obsolete.

athallas, 2015-12-06:
READMEs in [setup](https://github.com/rapp-project/rapp-platform/tree/master/rapp_scripts/setup) and [deploy](https://github.com/rapp-project/rapp-platform/tree/master/rapp_scripts/deploy) are not up-to-date. They should be updated or probably deleted (@etsardou ?) since they are included in the main [rapp_scripts README](https://github.com/rapp-project/rapp-platform/tree/master/rapp_scripts/README.md).

Additionally, in the deploy section `./deploy_hop_services` must be corrected. Finally, the screen reference should be made optional and different ttys for the two deploy scripts should be presented as the default option.

etsardou, 2015-12-06:
Indeed, they should be updated according to the wiki pages.

etsardou, 2015-12-07:
5cffe8328347453ef08031457ea8c466f186544f

closed 2015-12-07 by etsardou