rapp-project/rapp-platform #196: Travis: send mail to commiter if build fails

etsardou, 2015-12-07:


athallas, 2015-12-07:
As specified [here](https://docs.travis-ci.com/user/notifications/#How-is-the-build-email-receiver-determined%3F) the default behavior is to notify the commiter and the author. However, since in the [travis.yaml](https://github.com/rapp-project/rapp-platform/blob/master/.travis.yml) file another recipient has been specified, the default behavior is canceled and only the specified recipient is notified.

I would suggest removing the following lines from the travis configuration file:
```
notifications:
  email:
    recipients:
      - etsardou@gmail.com
    on_success: always #[always|never|change] # default: change
    on_failure: always #[always|never|change] # default: always
```

etsardou, 2015-12-07:
Indeed, we should change this in 0.5.5

commit 4798f949 (2015-12-08): Fixed travis notifications.

Travis now uses default notification settings, emailing the author and the
commiter. (closes #196)
closed 2015-12-08 by athallas