rapp-project/rapp-platform #198: Integrate creation of default rapp user on clean_install

klpanagi, 2015-12-07:
A script exists for rapp_user_creation: [create_rapp_user.sh](https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/devel/create_rapp_user.sh)

Use this script on clean_install to create the default rapp user.

Also:
- Delete the "**rapp**" user from [RappPlatformMySqlDbSchema](https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/setup/RappPlatformMySqlDbSchema.sql).
- Delete the "**rapp**" user from hop-authenticated-users, [here](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/config/hop/hoprc.js).

athallas, 2016-04-09:
@klpanagi What's the status? Should this be removed due to the new database?

klpanagi, 2016-04-09:
This is related to HttpBasicAuthentication previously used to authenticate users for accessing Platform's resources.
We currently use token-based application authentication mechanisms for these purposes so this issue os considered to be deprecated and should be closed.

closed 2016-04-09 by athallas