rapp-project/rapp-platform #200: clean_install.sh fails during RAPP Platform installation

dudekw, 2015-12-09:
Clean_install.sh fails during RAPP Platform installation.

Terminal listing:
```shell
 [RAPP] ROS - Setup sources list 
[sudo] password for rapp: 
 [RAPP] ROS - Setup keys 
 [RAPP] ROS - Installing 
 [RAPP] ROS - Initializing rosdep 
 [RAPP] Setup ROS environment 
 [RAPP] Installing auxiliary packages 
 [RAPP] Installing Rosjava 
 [RAPP] Cloning Rosjava 
 [RAPP] Building Rosjava 
 [RAPP] Installing Knowrob 
 [RAPP] Initializing Knowrob Catkin Workspace
 [RAPP] Cloning Knowrob 
 [RAPP] Building Knowrob 
 [RAPP] Knowrob Finished 
[sudo] password for rapp: 
 [RAPP] Create Github folders 
 [RAPP] Initializing Rapp Catkin Workspace
 [RAPP] Cloning the rapp-platform repo, branch: 
[Error]: RAPP Platform installation failed on rapp-platform setup
 ```

etsardou, 2015-12-09:
Please check your local branch of rapp_platform is synchronized with origin/master. This issue was fixed for clean_install 2-3 days ago [here](https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/setup/clean_install.sh#L60)

I am closing the issue. If it persists, please reopen it.

dudekw, 2015-12-09:
The rapp-platform repository was cloned today.  The only update after git pull:

```shel
From https://github.com/rapp-project/rapp-platform
 + 3c60a23...ac153bf cognitive_system -> origin/cognitive_system  (forced update)
Already up-to-date.
```
6_rapp_platform_setup.sh script works fine.

etsardou, 2015-12-09:
Can you please check your clean_install.sh file and see if these exist in lines 60-63:

```
RAPP_PLATFORM_BRANCH='master'
if [ -n "${TRAVIS_BRANCH}" ]; then
  RAPP_PLATFORM_BRANCH="${TRAVIS_BRANCH}"
fi
```

If yes something strange is going on...

etsardou, 2015-12-09:
Well yes, this is indeed a bug! Thanks

etsardou, 2015-12-09:
Master branch is fixed now (a40538673cf47e66aa432f7529b845e368c0dcc6). I have opened a new issue #201 and we will check it again in a clean VM. Thanks again!

closed 2015-12-09 by etsardou

dudekw, 2015-12-09:
I'm glad I could help!