rapp-project/rapp-platform #206: Ontology backup was not loaded.. Continuing with empty ontology

alexge233, 2015-12-11:
Hello,

Just did: 
```
git pull
rm -rf build/ devel/
catkin_make
```

Running the rapp_deploy script, I got:

    [ERROR] [1449862193.651784507]: Ontology backup was not loaded.. Continuing with empty ontology
    [ERROR] [1449862193.651966685]: File does not exist in provided file path: '/home/rapp/rapp_platform_files/currentOntologyVersion.owl'
    [ INFO] [1449862193.652081945]: KnowRob ROS wrapper initialized

Is this a path issue?

thanoski, 2015-12-11:
Hello Alex,
The error says File does not exist in provided file path: '/home/rapp/rapp_platform_files/currentOntologyVersion.owl'

Can you check if the file exists? Path looks good to me, assuming the username is rapp

etsardou, 2015-12-12:
As of v0.5.5 the knowrob ontology is automatically stored and retrieved from ```/home/rapp/rapp_platform_files/currentOntologyVersion.owl```. 

This error was shown the first time you launched the code, as you didn't have this file but now it has been probably created (@thanoski ? ). If it doesn't you can copy it from ```rapp/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp-scripts/setup/currentOntologyVersion.owl```

closed 2015-12-12 by etsardou

alexge233, 2015-12-12:
This error is reproduced every time I run the depoy_ros script.
Within the directory `$HOME/rapp_platform_files/` the file `currentOntologyVersion.owl` does not exist.
As per your instructions I setup a symbolic link to it, and it now works fine.

etsardou, 2015-12-12:
@thanoski shouldn't the file be created if it was not there the first time?

thanoski, 2015-12-12:
not necessarily.. the dump process is triggered (and as a result the file is created) whenever a write to the ontology takes place. 
But the file was supposed to be copied there by the rapp setup install scripts.
Anyway, just run the tests and the file will be created, the ontology tests will fail though..
run
catkin_make run_tests_rapp_knowrob_wrapper -j1
within the rapp platform catkin workspace folder

Bottom line: The file should have been placed there by the rapp install setup scripts.