rapp-project/rapp-platform #207: grunt and init-hop

alexge233, 2015-12-11:
Is there a new dependency using `grunt` to launch hop?
My copy of the RAPP VM has replaced
```
rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_scripts/deploy/deploy_hop_services.sh
```

with

```
rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_scripts/deploy/deploy_web_services.sh
```

When I try to execute it I get errors about grunt not being installed.

etsardou, 2015-12-12:
True, as of v0.5.5 the web scripts have a dependency on ```grunt```. Please execute these two commands:

https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/setup/3_auxiliary_packages_setup.sh#L43

@klpanagi can elaborate on the dependency.

closed 2015-12-12 by etsardou

klpanagi, 2015-12-12:
rapp_web_services has a dependency on Grunt.js since v0.5.5.
In order to use Grunt.js for task management, you will first need to install, **system-wide**, grunt-commandline-interface:

```shell
$ sudo npm install -g grunt-cli
```

or just execute the script @etsardou  mentioned.

alexge233, 2015-12-12:
At which port does grunt deploy HOP?
I installed grunt, and setup the env variables, and I get no output from the script or from HOP.
Is this the defined behaviour?
Port 9001 is not open btw.

alexge233, 2015-12-12:
Also, looking into the script I realise it does:

    WEB_SERVICES_PATH=$HOME/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services/                                    
    cd  $WEB_SERVICES_PATH                                                                                                                
    grunt --verbose init-hop 

However, there is no `init-hop` script in that directory.

klpanagi, 2015-12-12:
This is really strange. I tried to reproduce this error with no success.

- There is no redirection, either of stdout or stderror so what do you mean the script silently fails?
- Did you install the RAPP Platform using the [clean_install.sh](https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/setup/clean_install.sh) script?

Please follow these steps and report the output:
- Make sure your local branch is synchronized with the origin and you are at the master branch:

```shell
$ cd ~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform
$ git checkout origin/master --track
$ git fetch && git reset --hard origin/master
$ git checkout -- .
```
-  Navigate to the [rapp_web_services package](https://github.com/rapp-project/rapp-platform/tree/master/rapp_web_services), cleanup and install depended packages:

```shell
$ cd ~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services
$ make clean
$ npm install
```

- Run the, **init-hop** grunt task:

```shell
$ grunt init-hop
```

If the aforementioned grunt task fails for any reason, please execute the [run.sh](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/run.sh) script,  directly:

```shell
$ ./run.sh
```

Please report back the output of the above commands.

alexge233, 2015-12-13:
@klpanagi I did exactly what you described above:

```
- cd rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform] (master *)
- git checkout origin/master --track
fatal: A branch named 'master' already exists.
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform] (master *)
- git fetch && git reset --hard origin/master
HEAD is now at dcf0daa fixes #202
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform] (master)
- git checkout -- .
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform] (master)
- cd rapp_web_services/
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services] (master)
- make clean
rm -rf doc/
rm -rf node_modules
[rapp@rapp-platform][~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services] (master)
- npm install
npm WARN package.json @ No repository field.
npm http GET https://registry.npmjs.org/grunt-contrib-watch
npm http GET https://registry.npmjs.org/grunt-contrib-nodeunit
npm http GET https://registry.npmjs.org/grunt
npm http GET https://registry.npmjs.org/grunt-contrib-jshint
npm http GET https://registry.npmjs.org/grunt-shell
npm http GET https://registry.npmjs.org/ink-docstrap
npm http GET https://registry.npmjs.org/grunt-jsdoc
npm http 304 https://registry.npmjs.org/grunt
npm http 304 https://registry.npmjs.org/grunt-shell
npm http 304 https://registry.npmjs.org/ink-docstrap
npm http 200 https://registry.npmjs.org/grunt-contrib-watch
npm http 304 https://registry.npmjs.org/grunt-jsdoc
npm http GET https://registry.npmjs.org/grunt-contrib-watch/-/grunt-contrib-watch-0.6.1.tgz
npm http 200 https://registry.npmjs.org/grunt-contrib-watch/-/grunt-contrib-watch-0.6.1.tgz
npm http 304 https://registry.npmjs.org/grunt-contrib-jshint
npm http GET https://registry.npmjs.org/grunt-contrib-jshint/-/grunt-contrib-jshint-0.10.0.tgz
npm http 200 https://registry.npmjs.org/grunt-contrib-jshint/-/grunt-contrib-jshint-0.10.0.tgz
npm http 200 https://registry.npmjs.org/grunt-contrib-nodeunit
npm http GET https://registry.npmjs.org/grunt-contrib-nodeunit/-/grunt-contrib-nodeunit-0.4.1.tgz
npm http 200 https://registry.npmjs.org/grunt-contrib-nodeunit/-/grunt-contrib-nodeunit-0.4.1.tgz
npm http GET https://registry.npmjs.org/nodeunit
npm http GET https://registry.npmjs.org/hooker
npm http GET https://registry.npmjs.org/jshint
npm http GET https://registry.npmjs.org/chalk
npm http GET https://registry.npmjs.org/gaze
npm http GET https://registry.npmjs.org/tiny-lr-fork/0.0.5
npm http GET https://registry.npmjs.org/lodash
npm http GET https://registry.npmjs.org/async
npm http GET https://registry.npmjs.org/jsdoc
npm http GET https://registry.npmjs.org/async
npm http GET https://registry.npmjs.org/coffee-script
npm http GET https://registry.npmjs.org/colors
npm http GET https://registry.npmjs.org/dateformat/1.0.2-1.2.3
npm http 304 https://registry.npmjs.org/hooker
npm http GET https://registry.npmjs.org/eventemitter2
npm http GET https://registry.npmjs.org/findup-sync
npm http GET https://registry.npmjs.org/glob
npm http GET https://registry.npmjs.org/minimatch
npm http 304 https://registry.npmjs.org/jshint
npm http GET https://registry.npmjs.org/lodash
npm http GET https://registry.npmjs.org/iconv-lite
npm http GET https://registry.npmjs.org/rimraf
npm http 200 https://registry.npmjs.org/tiny-lr-fork/0.0.5
npm http GET https://registry.npmjs.org/nopt
npm http GET https://registry.npmjs.org/underscore.string
npm http GET https://registry.npmjs.org/js-yaml
npm http GET https://registry.npmjs.org/getobject
npm http GET https://registry.npmjs.org/grunt-legacy-util
npm http 304 https://registry.npmjs.org/chalk
[... 551 lines cut ...]
-13 WITH-URL: http://rapp-platform:9001/hop/text_to_speech
12 REQUEST.serv: GET *://rapp-platform:9001 /hop/text_to_speech 
12 ERROR: #<JsError: TypeError, toObject: cannot convert "undefined">
File "services/text_to_speech.service.js", line 111, character 28:
#  var unqCallId = randStrGen.createUnique();
#                            ^
TypeError: toObject: cannot convert "undefined"
    1. \@\@\@hopscript, services/text_to_speech.service.js:111
    2. &thread2229

 ---> [Failed]: text_to_speech

 <UpRunning Services>
    *[0] - available_services

------------------------------------------------------------
------------------------------------------------------------

13 ABORT:  &io-connection-error
*** ERROR:read/timeout:
Connection reset by peer -- #<input_port:#unspecified.512>
    1. &pool-scheduler1352, 
```

Please note that I have been using the RAPP VM, without any modifications whatsoever!

etsardou, 2015-12-13:
To clarify, you are using the old VM (0.4.1) or the new (0.5.5)?

alexge233, 2015-12-13:
FYI, indeed the path
`~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services/modules/RosBridgeJS`
is empty and does not contain anything.

alexge233, 2015-12-13:
I am going take a wild guess and say the 0.4.1. Since I have been working on it for months.
Its the VM that is broken?

etsardou, 2015-12-13:
No its not broken, its a submodules problem. Please erase the ```~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform``` folder and execute there:

```
git clone --recursive https://github.com/rapp-project/rapp-platform.git
```

Then it will be fine (after compilation of course).

etsardou, 2015-12-13:
The problem was that the RosBridgeJS submodule did not exist in the previous rapp-platform and the ```git fetch``` did not bring it. One way to get it is by recursive cloning like above.

alexge233, 2015-12-13:
@etsardou  I did what you suggested (and rebuild).
The `grunt` command still does not work.
Executing `run.sh` from inside the `rapp_web_services` directory:

```
- ./run.sh 
Hop configuration parameters passed:
  * Http-Port: 9001
  * Server Cache directory: /home/rapp/.hop/cache/server
  * Clear cache directory: true
  * Server log directory: /home/rapp/.hop/log/server
  * Scheduler policy: accept-many
  * Max threads: 100
  * Report execution time: false
  * Verbosity level: 1
  * Debug level: 1
  * Warning level: 1
  * Security level: 0

Initialing Hop Web Server and registering Web Services
+ hop -v1 -g1 -w1 -s0 --clear-cache --no-fast-server-event --cache-dir /home/rapp/.hop/cache/server --http-port 9001 --log-file /home/rapp/.hop/log/server/hop-server-201512132118.log --scheduler accept-many --max-threads 100 --rc-file ./config/hop/hoprc.hop ./services/init.js
Hop v3.0.0
  http: 9001
  hostname: rapp-platform
  hostip: 127.0.1.1
...logging to [/home/rapp/.hop/log/12-13-2015-1450034322144/]

...file cache directory is set at [/home/rapp/.hop/cache/services/]

...file cache directory is set at [/home/rapp/.hop/cache/server/9001/upload/]

Found hop service file: [ontology_superclasses_of.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/ontology_superclasses_of.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/ontology_superclasses_of
Found hop service file: [ontology_is_subsuperclass_of.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/ontology_is_subsuperclass_of.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/ontology_is_subsuperclass_of
Found hop service file: [qr_detection.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/qr_detection.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/qr_detection
Found hop service file: [face_detection.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/face_detection.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/face_detection
Found hop service file: [speech_detection_sphinx4.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/speech_detection_sphinx4.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/speech_detection_sphinx4
Found hop service file: [text_to_speech.service.js]
http://rapp-platform:9001/hop/ontology_subclasses_of
Found hop service file: [record_cognitive_test_performance.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/record_cognitive_test_performance.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/record_cognitive_test_performance
Found hop service file: [rapp_platform_status.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/rapp_platform_status.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/rapp_platform_status
Found hop service file: [speech_detection_google.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/speech_detection_google.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/speech_detection_google
Found hop service file: [set_noise_profile.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450034322144/set_noise_profile.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/set_noise_profile
Found hop service file: [cognitive_test_chooser.service.js]
[... 114 lines cut ...]
    1. \@\@\@@loopInf hopscript, services/rapp_platform_status.service.js:620
    2. &thread2384
-17 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
16 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
TypeError: no such field "services" pair
    1. \@\@\@@loopInf hopscript, services/rapp_platform_status.service.js:620
    2. &thread2384
-18 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
17 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
TypeError: no such field "services" pair
    1. \@\@\@@loopInf hopscript, services/rapp_platform_status.service.js:620
    2. &thread2384
-19 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
18 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
TypeError: no such field "services" pair
    1. \@\@\@@loopInf hopscript, services/rapp_platform_status.service.js:620
    2. &thread2384
-20 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
19 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
TypeError: no such field "services" pair
    1. \@\@\@@loopInf hopscript, services/rapp_platform_status.service.js:620
    2. &thread2384
```

Seems like it gets stuck in an infinite loop (the output kept repeating on and on)

etsardou, 2015-12-13:
The services are working, you just get this error for a single service (which is not needed). I think this is due to the HOP version you have -- I got this too, uninstalled HOP and installed it again and worked. 

Anyway, the services are live now and you can work. @klpanagi can tell you what to do (if it's not HOP's fault) to eliminate this message.

alexge233, 2015-12-13:
@etsardou So I should stick with `run.sh` for now instead of `grunt`?
@klpanagi should I uninstall HOP and re-install, or can I just edit that infinite loop out?

etsardou, 2015-12-13:
You should always use ```deploy_wab_services.sh``` from the rapp_scripts folder in rapp_platform to deploy the services.

alexge233, 2015-12-13:
@etsardou It still fails, only `run.sh` seems to work. It must be an issue with grunt.

etsardou, 2015-12-13:
Oh this calls grunt as well! Sorry I though it invoked run.sh. Then waiting for @klpanagi ...

klpanagi, 2015-12-13:
The errors you get is due to the old version of hop+bigloo you have.

Please install hop+bigloo by executing the [11_hop_setup.sh](https://github.com/rapp-project/rapp-platform/blob/master/rapp_scripts/setup/11_hop_setup.sh) script. Make sure to remove hop+bigloo directories before:

```shell
rm -rf ~/rapp_platform/hop*
cd ~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_scripts/setup
./11_hop_setup.sh
```

The above script will fetch and install the right versions of hop+bigloo for v0.5.5.

As for the grunt command, if you take a closer look at the [init-hop](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/Gruntfile.js#L39) task, you will notice that it just executes [run.sh](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/run.sh) script on current shell, while keeping stdout and stderr active too and that is why the reported issue seems weird.
I will dig into it tomorrow.

Can you please check on the installed grunt-cli version:

```shell
npm list -g grunt-cli
```

You should see:

``` shell
/usr/local/lib
└── grunt-cli@0.1.13
```

etsardou, 2015-12-13:
Actually first uninstall hop and bigloo with ```sudo make uninstall``` in the ```~/rapp_platform/hop*``` folders and then proceed with the other steps..

klpanagi, 2015-12-13:
Check on node.js version too.

```shell
node --version
```

alexge233, 2015-12-13:
npm:

```
- npm list -g grunt-cli
/usr/local/lib
└── grunt-cli@0.1.13 

- npm list -g grunt-cli
/usr/local/lib
└── grunt-cli@0.1.13 
```

I just finished running `./11_hop_setup.sh `
```
- ./run.sh 
Hop configuration parameters passed:
  * Http-Port: 9001
  * Server Cache directory: /home/rapp/.hop/cache/server
  * Clear cache directory: true
  * Server log directory: /home/rapp/.hop/log/server
  * Scheduler policy: accept-many
  * Max threads: 100
  * Report execution time: false
  * Verbosity level: 1
  * Debug level: 1
  * Warning level: 1
  * Security level: 0

Initialing Hop Web Server and registering Web Services
+ hop -v1 -g1 -w1 -s0 --clear-cache --no-fast-server-event --cache-dir /home/rapp/.hop/cache/server --http-port 9001 --log-file /home/rapp/.hop/log/server/hop-server-201512132328.log --scheduler accept-many --max-threads 100 --rc-file ./config/hop/hoprc.hop ./services/init.js
Hop v3.0.0
  http: 9001
  hostname: rapp-platform
  hostip: 127.0.1.1
...logging to [/home/rapp/.hop/log/12-13-2015-1450042124529/]

...file cache directory is set at [/home/rapp/.hop/cache/services/]

...file cache directory is set at [/home/rapp/.hop/cache/server/9001/upload/]

Found hop service file: [ontology_superclasses_of.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/ontology_superclasses_of.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/ontology_superclasses_of
Found hop service file: [ontology_is_subsuperclass_of.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/ontology_is_subsuperclass_of.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/ontology_is_subsuperclass_of
Found hop service file: [qr_detection.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/qr_detection.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/qr_detection
Found hop service file: [face_detection.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/face_detection.log]
[OK] - Initiated HOP Web Service: 
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/text_to_speech
Found hop service file: [ontology_subclasses_of.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/ontology_subclasses_of.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/ontology_subclasses_of
Found hop service file: [record_cognitive_test_performance.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/record_cognitive_test_performance.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/record_cognitive_test_performance
Found hop service file: [rapp_platform_status.service.js]
Initiated Writable Stream [/home/rapp/.hop/log/12-13-2015-1450042124529/rapp_platform_status.log]
[OK] - Initiated HOP Web Service: 
http://rapp-platform:9001/hop/rapp_platform_status
Found hop service file: [speech_detection_google.service.js]
[... 156 lines cut ...]
33 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-35 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
34 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-36 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
35 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-37 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
36 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-38 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
37 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-39 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
38 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-40 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
39 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&vals=. 
-41 WITH-HOP http://localhost:9001/hop/available_services?hop-encoding=hop&vals=.
40 REQUEST.serv: PUT *://localhost:9001 /hop/available_services?hop-encoding=hop&va
```

For some reason `QR-Detection`, `Face-Detection` and `Set-Noise-Profile` produce:
```
Service invocation error. Invalid {file_uri} field! Abortion for security reasons.
```

Why is this?

Also how can I disable the infinite loop of the `/hop/available_services?hop-encoding=hop&vals=.`???

klpanagi, 2015-12-14:
The "infinite loop" is due to the existence of the rapp_platform_status HOP Service. This will be deprecated in v0.6.0.
As for now you can just disable registration of the rapp_platform_status service:

```shell
$ cd ~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services/services
$ mv rapp_platform_status.service.js rapp_platform_status.js
```

Any updates on

```shell
$ grunt init-hop
```
??

alexge233, 2015-12-14:
@klpanagi 

`$ grunt init-hop` still doesn't work for me. No output, hop isn't brought up.
I'll disable the status service, thank you very much for your help.