rapp-project/rapp-platform #225: [Front-End]: cognitive_get_scores service per-exercise queries

klpanagi, 2015-12-27:
[cognitive_get_scores](https://github.com/rapp-project/rapp-platform/blob/front-end/rapp_web_services/services/cognitive_get_scores.service.js#L190) service should allow per-exercise queries.

@akintsakis  Is this currently supported by the cognitive_system RIC nodes?

akintsakis, 2015-12-27:
No it does not, it only returns per test type (Arithmetic, Reasoning, Recall) scores.

klpanagi, 2015-12-27:
That's what i meant.

We should be able to retrieve performance scores per test-type

akintsakis, 2015-12-27:
ok, I will apply the necessary modifications in the respective Cognitive exercise ROS service

klpanagi, 2016-01-02:
Altered Service response objects as of 0333be4f4bea5b3896ca99bba063968f166d0293

closed 2016-01-02 by klpanagi