rapp-project/rapp-platform #226: [Front-End]: cognitive_get_history service per-exercise queries

klpanagi, 2015-12-27:
[cognitive_get_history](https://github.com/rapp-project/rapp-platform/blob/front-end/rapp_web_services/services/cognitive_get_history.service.js#L190) service should allow per-exercise queries.

@akintsakis  Is this currently supported by the cognitive_system RIC nodes?

akintsakis, 2015-12-27:
When designing the service we decided to include all records and then filtering should take place to narrow it down to the desirable subset. All necessary information (ie test type, test sub type, individual test scores ) is contained in the service repsonse

klpanagi, 2015-12-27:
Should be a better approach for the web-service to allow an optional input argument for clients to retrieve  per-testType information.

We can integrate a simple parser under the Web-Service for this purposes.

Though either way response objects should be always static.

akintsakis, 2015-12-27:
Yes, we had decided for the filtering to take place in the hop web service. So the hop web service that communicates with the cognitive_get_scores ROS service should be modified accordingly. The ROS service does not need to be modified. The changes in the hop service are trivial, just loop through the results and keep only the desired. Do you want to do it, or should I do it?

klpanagi, 2016-01-02:
Altered Service response objects as of 0333be4f4bea5b3896ca99bba063968f166d0293

closed 2016-01-02 by klpanagi