rapp-project/rapp-platform #227: [Cognitive-System]: Scores not in range [0, 100]

klpanagi, 2015-12-27:
Total user's performance score on cognitive exercises should be in range [0, 100].
Though i get a response from cognitive_get_scores service that exceeds this range

```python
{u'error': u'', u'testCategories': [u'ArithmeticCts', u'ReasoningCts', u'AwarenessCts'], u'testScores': [16, 150, 50]}
```

akintsakis, 2015-12-27:
When recording a new cognitive test performance entry, the score is in range [0,100]. These scores are different, they are cumulative and multipliers are applied based on the score's difficulty level, thus they are not restricted in the [0,100] range.

closed 2015-12-27 by akintsakis

klpanagi, 2015-12-27:
So i understand that there does not exist a normalized range for global performance scores.
Is this right?

klpanagi, 2016-01-02:
Score values should be now normalized in range [0, 100].
Functional tests should be created to evaluate this.