rapp-project/rapp-platform #233: [HOP services]: return type of `text_to_speech`

alexge233, 2016-01-29:
boost and C++ has no native base64 decoding scheme, can you please return raw audio data, as an HTTP multiform/part:

- return the JSON in a multiform/data: 
`{ payload: <data_boundary>, basename: <audio_file_basename>, encoding: <payload_encoding>, error: <error_message> }`
- return the actual audio data, separated by the multiform.

Example:
```
--(random_boundary)\r\n
{JSON HERE}\r\n
--(random_data_boundary)\r\n
[Raw PCM16/WAV Data Here]
```

Please make sure the JSON `payload` references the `random_data_boundary` of the audio data so that we can retrieve it.

klpanagi, 2016-01-29:
As previously stated, we decided to keep client-server interfaces simple.
We defined client-server interfaces to be **only** json formatted objects.

Also through HOP Server there is currently no support for crafting complex response bodies, so we cannot support this approach.

You can use **openssl** cross-platform library to encode/decode data.
[openssl_base64](https://www.openssl.org/docs/manmaster/crypto/BIO_f_base64.html)

OpenSSL's encoding to base64 is binary safe.
[OpenSSL Base64 Encoding: Binary Safe](https://gist.github.com/barrysteyn/7308212)

alexge233, 2016-01-29:
You can use Node.JS for this, its literally 5 lines of code:
To craft the reply form:

```
var form = new formData();
form.append('file_uri', fs.createReadStream(audio), { filename: filename });
form.append('audio_source', audio_source);
```

And then simply attach the form to the header, using `http.ServerResponse`
Using base64 violates the whole point of using [RFC 2388 multipart/form-data](https://www.ietf.org/rfc/rfc2388.txt) for which we went through the trouble of implementing the entire API (C++/JS).

Also, I don't wanna add any kind of dependency other than boost (which is header only),
because we will be porting the API to ARM and Windows, making it very difficult if more dependencies are attached to it.

Your JSON reply does not have to be changed, simply reference the boundary.

If you do not want to reference the boundary, then reply with JSON & raw data, and make sure only one audio file per HTTP transaction is attached, we will extract the first boundary as JSON, and the second as RAW audio.

alexge233, 2016-01-29:
Freeze this issue for now, I can implement b64 enc/decoding with boost only,
but this is not the correct approach of transmitting raw data, and should eventually be resolved using proper multipart/form-data.

Have a look at the JS API if want to see how you can use Node.JS, although I don't know if HOP will accept Node.JS http responses.

klpanagi, 2016-01-29:
> You can use Node.JS for this, its literally 5 lines of code:

RAPP Platform front-end is based on HOP Server (Not node.js http server) and it is developed using [Hop.js](http://hop.inria.fr/home/index.html) and not native Node.js.

> Using base64 violates the whole point of using RFC 2388 multipart/form-data for which we went through the trouble of implementing the entire API (C++/JS).

We went through this regarding **posting** to HOP Server using [RFC multipart/form-data](https://www.ietf.org/rfc/rfc2388.txt), which is currently supported.

Currently Hop.js supports [these](http://hop.inria.fr/home/00-hop.html#responses) service responses.

> this is not the correct approach of transmitting raw data, and should eventually be resolved using proper multipart/form-data.

I absolutely agree. We can contact Hop.js developers and ask for a patch on this so we can properly create custom responses.

closed 2016-01-29 by alexge233