rapp-project/rapp-platform #234: Suggestion to rename `speech_detection` to `speech_recognition`

alexge233, 2016-01-31:
I would suggest we rename `speech_detection_google` and `speech_detection_sphinx` to `speech_recognition_X`.

The term `speech_detection` implies we're detecting speech, as in *diarization* which is a feature I would like to implement either locally in the API, or as a Service. It's purpose is to detect audio speech within audio files, whereas the Google and Sphinx4 API are recognizing speech as text.

etsardou, 2016-02-01:
Indeed, you are right. If you plan to implement this in the near future, instead of ```speech detection``` use the more appropriate term ```voice activity detection```.

We will plan this change for the 0.7 release.

alexge233, 2016-02-02:
How about `diarization` which is the keyword normally used?
I've found: [shout](http://shout-toolkit.sourceforge.net/download.html) which is a continuous diarization written in C++ as somebody's PhD research.

etsardou, 2016-02-03:
Sure, as you prefer.