rapp-project/rapp-platform #235: speech_detection_sphinx4 JSGF grammar parameter

alexge233, 2016-01-31:
Why is the parameter `grammar` an array in `speech_detection_sphinx4`?
Have you enabled loading more than one grammar rule?
Can somebody please verify this? 
I think that it should be one string (containing the entire BNF/JSGF text).

etsardou, 2016-02-01:
Currently the ```grammar``` parameter supports the ARPA-based model [ref](http://cmusphinx.sourceforge.net/wiki/tutoriallm#arpa_model_training), thus each element of the list is a sentence which must contain only words included in the ```words``` parameter.

In order for the BNF/JSGF rules to be supported, significant changes must occur in the node.

Thus I am changing the title, in order to implement this feature at the future.

alexge233, 2016-02-02:
I would actually prefer to create an API extension which uses pocketsphinx in C++, and wrap it for *JSGF-only* usage.
The same code could be used on the cloud (although it would make little sense to me).
Regarding this issue I'm closing it, but we should update the documentation to state its not JSGF.

closed 2016-02-02 by alexge233