rapp-project/rapp-platform #236: Microsoft CNTK: deep-learning toolkit

klpanagi, 2016-02-01:
Microsoft has released open-source it's deep-learning toolkit:
[Microsoft-CNTK](https://github.com/Microsoft/CNTK)

alexge233, 2016-02-02:
Will this build on Unix/Linux?
I would prefer usage of JavaScript based related services, which could be easily wrapped around with node.JS:

* [npm dnn](https://www.npmjs.com/package/dnn)
* [conv net js](https://github.com/karpathy/convnetjs)
* [recurrent js](https://github.com/karpathy/recurrentjs)
* [brain js](https://github.com/harthur/brain)
* [synaptic js](https://github.com/cazala/synaptic)

If you have something specific in mind (e.g., object recognition) we would then require GPGU capabilities, and we could use [caffee](https://github.com/rapp-project/caffe) which somebody has forked for rapp.

Using MS code, even on Unix/Linux could be very tricky.

etsardou, 2016-02-03:
This issue is just for us having it in mind, it's not meant for implementation for now. Concerning the caffe, Thanos is already working on a wrapper (that's why it is forked).

closed 2016-03-07 by etsardou