rapp-project/rapp-platform #262: deploy_web_services.sh fails with error: Maximum call stack size exceeded

maciek-slon, 2016-03-14:
Platform compiled from sources today. Branch devel.

Call:
```
$ ./deploy_web_services.sh 
```

Result:
```
Running "shell:init_hop" (shell) task
Hop configuration parameters passed:
  * Http-Port: 9001
  * Server Cache directory: /home_local/rapp/.hop/cache/server
  * Clear cache directory: true
  * Server log directory: /home_local/rapp/.hop/log/server
  * Scheduler policy: accept-many
  * Max threads: 100
  * Report execution time: false
  * Verbosity level: 1
  * Debug level: 1
  * Warning level: 1
  * Security level: 0

Initialing Hop Web Server and registering Web Services
+ hop -v1 -g1 -w1 -s0 --clear-cache --no-fast-server-event --cache-dir /home_local/rapp/.hop/cache/server --http-port 9001 --log-file /home_local/rapp/.hop/log/server/hop-server-201603141639.log --scheduler accept-many --max-threads 100 --rc-file ./config/hop/hoprc.hop ./init.js
Hop v3.0.0
  zeroconf: avahi 0.6.31
  http: 9001
  hostname: roman.robotyka.ia.pw.edu.pl
  hostip: 192.168.18.171
error: uncaughtException: Maximum call stack size exceeded 
{ date: 'Mon, 14 Mar 2016 16:39:34 +0100',
  process: 
   { pid: 13119,
     uid: 1001,
     gid: 1001,
     cwd: '/home_local/rapp/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_web_services',
     execPath: '/usr/local/bin/hop',
     version: '3.0.0',
     argv: 
      [ 'hop',
        '-v1',
        '-g1',
        '-w1',
        '-s0',
        '--clear-cache',
        '--no-fast-server-event',
        '--cache-dir',
        '/home_local/rapp/.hop/cache/server',
        '--http-port',
        '9001',
        '--log-file',
        '/home_local/rapp/.hop/log/server/hop-server-201603141639.log',
        '--scheduler',
        'accept-many',
        '--max-threads',
        '100',
        '--rc-file',
        './config/hop/hoprc.hop',
        './init.js' ],
     memoryUsage: 
      { rss: 98820096,
        heapTotal: 0,
        heapUsed: 0 } },
  os: 
   { loadavg: 
      { buffer: { byteLength: 24, slice: [Function: slice] },
        BYTES_PER_ELEMENT: 8,
        length: 3,
        byteLength: 24,
        byteOffset: 0,
        set: [Function: set],
        get: [Function: get],
        subarray: [Function: subarray] },
     uptime: 5492 },
  trace: [],
  stack: undefined }
+ set +o xtrace

Done, without errors. (Yeeeeeeah, rrrrrright ;-)
```

maciek-slon, 2016-03-14:
Hmmm, I think I've just fixed this error. In some places, /home was hardcoded. As on my current machine /home folder is unavailable and /home_local is used instead, scripts crashed on startup. I've pushed fixed files directly to devel branch.

closed 2016-03-14 by maciek-slon

klpanagi, 2016-03-15:
Thank you @maciek-slon  for pointing out this issue.

commit 06766eb0 (2016-03-14): Change hardcoded '/home' in JS startup scripts. Fix #262.

Hardcoded /home was changed to proper home directory of user.
- ${HOME} in shell scripts
- process.env.HOME in JavaScript files
commit cf5ceed7 (2016-03-23): Change hardcoded '/home' in JS startup scripts. Fix #262.

Hardcoded /home was changed to proper home directory of user.
- ${HOME} in shell scripts
- process.env.HOME in JavaScript files