rapp-project/rapp-platform #279: Path planning integration tests fail

athallas, 2016-04-08:
This is due to an error in the path of `map_path` variable in: 

`~/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_path_planning/rapp_path_planning/src/path_planning.cpp` 

You can see in line 262:

```cpp
std::string map_path = homedir_str+"/rapp_platform_files/maps/rapp/"+req.user_name+"/"+req.map_name+".yaml";
```

This could most probably be the result of file path inconsistencies due to functional/unit test created files that were not deleted.

Please start a new branch from `devel` to fix the issue and validate all unit, functional and integration tests in their own merit, i.e. delete the `homedir_str+"/rapp_platform_files/maps` after each test.

klpanagi, 2016-04-08:
We previously specified that maps will be stored under
```shell
'${HOME}/rapp_platform_files/maps/<user>/
```

So this:
```cpp
std::string map_path = homedir_str+"/rapp_platform_files/maps/rapp/"+req.user_name+"/"+req.map_name+".yaml";
```

should be resolved to
```cpp
std::string map_path = homedir_str+"/rapp_platform_files/maps/"+req.user_name+"/"+req.map_name+".yaml";
```

etsardou, 2016-04-11:
@dudekw any updates?

commit da634eb0 (2016-04-11): Fixed path_planning failing tests (closes #279)
closed 2016-04-12 by athallas