rapp-project/rapp-platform #285: Add more integration tests

etsardou, 2016-04-12:


etsardou, 2016-05-17:
@klpanagi @athallas give a description / guideline on what integration tests we want

klpanagi, 2016-05-18:
**Integration tests**:  Test communication paths between different parts of the system done by the test to show that all modules/components (independent or cooperative) work correctly together.

**Communication paths between major components of the RAPP ecosystem**:
- Client (rapp-platform-api) ---> WebServices
- WebServices ---> ROS-Services
- ROS-Services ---> WebServices
- WebServices ---> Client (rapp-platform-api)

**Expected failures should be tested too**:
- Sending/Receiving invalid messages:
  * Attribute Errors
  * Type Errors
  * Value Errors
- Service existence
- Authentication failures
- Server Internal Errors
- Client Errors
- Server HTTP Errors (4xx)
- Client HTTP Errors (5xx)

closed 2016-07-01 by athallas