rapp-project/rapp-platform #297: weather_report_forecast and weather_report_current services do not exist

lpenteri, 2016-05-06:
I am calling the weather_report_forecast and weather_report_current services through the JS API using the following arguments: 'Athens', '', 0.
I get an error respond RAPP Platform Failure and after checking at the deploy_web_services and deploy_rapp_ros scripts' output I see the follows errors: 
"[Rosbridge]: Service /rapp/rapp_weather_reporter/weather_forecast does not exist" in the first, and
"[ERROR] [WallTime: 1462524988.658311] [Client 8] [id: Zeu5Uloux1] call_service InvalidServiceException: Service /rapp/rapp_weather_reporter/weather_forecast does not exist" in the second.

etsardou, 2016-05-06:
@athallas @klpanagi ?

klpanagi, 2016-05-07:
The log clearly states non existence of the weather_report* ROS Services on the machine hosting the RAPP Platform.

The weather_report node(s) have a dependency to [yweather](https://pypi.python.org/pypi/yweather/) so if you installed the RAPP Platform from a branch other than **devel** those dependencies have not been installed.
Please launch the rapp-platform and check the trace log for success execution of the weather_report node. In case of yweather is missing from the system, install it manually through the python package manager (pip):

```shell
sudo pip -H install yweather
```

On a clean Platform installation (on a personal machine), please make sure all tests pass.

- Execute unit and functional tests:

```shell
$ cd ${HOME}/rapp_platform/rapp-platform-catkin-ws && catkin_make run_tests
```

- Execute integration tests:

```shell
$ rosrun rapp_testing_tools rapp_run_tests.py
```

etsardou, 2016-05-08:
@lpenteri  Is this issue resolved?

lpenteri, 2016-05-09:
Indeed yweather was missing from the system. I installed it using this command:
`sudo -H pip install yweather`

Now when i call weather_report_current for example, I get:
weather_report_current JSON error: HTTPError()

alongside with an empty weather_current object: 
{
  "date": "",
  "temperature": "",
  "weather_description": "",
  "humidity": "",
  "visibility": "",
  "pressure": "",
  "wind_speed": "",
  "wind_temperature": "",
  "wind_direction": ""
}

On the output of the deploy_rapp_ros.sh script I see the following:
[ERROR] [WallTime: 1462801366.832976] [ FILE: yweather_reporter.py, FUNCTION: _fetch_yweather_report, LINE: 71 ]: HTTP Error 502: Connection refused

I also executed the tests. On the unit and functional tests most things were successful except from the following errors:

[ERROR] [WallTime: 1462801598.541165] [ FILE: speech_recognition_sphinx4.py, FUNCTION: speechRecognitionBatch, LINE: 138 ]: Word mine is not in words but exists in grammar
Word mine is not in words but exists in grammar
[ERROR] [WallTime: 1462801598.629657] [ FILE: speech_recognition_sphinx4.py, FUNCTION: speechRecognitionBatch, LINE: 138 ]: ERROR: Word ναι does not exist in the English Dictionary
ERROR: Word ναι does not exist in the English Dictionary
[ERROR] [WallTime: 1462801598.741645] [ FILE: speech_recognition_sphinx4.py, FUNCTION: speechRecognitionBatch, LINE: 138 ]: Word mine is not in words but exists in a sentence
Word mine is not in words but exists in a sentence

[ERROR] [WallTime: 1462801624.990465] [ FILE: email_sender.py, FUNCTION: sendEmailSrvCallback, LINE: 72 ]: Failed to handle attachment: 0qiyaexedmfgjacq
[ERROR] [WallTime: 1462801624.992286] [ FILE: email_sender.py, FUNCTION: sendEmailSrvCallback, LINE: 73 ]: Filename does not exist! 0qiyaexedmfgjacq
[ERROR] [WallTime: 1462801626.623818] [ FILE: email_sender.py, FUNCTION: _connectAndSend, LINE: 123 ]: SMTP failed!
[ERROR] [WallTime: 1462801626.627683] [ FILE: email_sender.py, FUNCTION: _connectAndSend, LINE: 124 ]: (535, '5.7.8 Username and Password not accepted. Learn more at\n5.7.8  https://support.google.com/mail/answer/14257 jd4sm31223928wjb.43 - gsmtp')

These ones are probably due to the same reason I wasn't able to use the email services with my gmail account, with gmail replying to me asking to use an app with more modern security standards. I will open a separate issue for this.

Lastly:
[ERROR] [1462801724.937709148]: [setParam] Failed to contact master at [devel-VirtualBox:37533].  Retrying...
[ERROR] [1462801726.657131907]: [setParam] Failed to contact master at [devel-VirtualBox:37533].  Retrying...
[ERROR] [1462801728.263443977]: [setParam] Failed to contact master at [devel-VirtualBox:37533].  Retrying...
[ERROR] [1462801729.753121467]: Failed to get a plan.
[ERROR] [1462801729.789773215]: Input algorithm does not exist
[ERROR] [1462801729.823956492]: Input map does not exist
[ERROR] [1462801729.858923534]: Input robot_type does not exist

About the integration tests there is no such file as rapp_run_tests.py anywhere in the 
${HOME}/rapp_platform/rapp-platform-catkin-ws
directory nor its subdirectories, so I did not manage to run them.

Is any of those expected or known bugs or did things go wrong with my installation? I installed the devel branch on a clean VM with the latest Ubuntu 14.04 OS and no errors were shown at the time of installation. Apart from that I applied a patch from the "https_fix" branch of the RAPP Platform repo later, in order for the TLSv1_2 to work properly with the authentication services.

klpanagi, 2016-05-09:
> I also executed the tests. On the unit and functional tests most things were successful except from the following errors:

Did you get a "TEST FAILED" report on the logs? Some tests are expected to produce errors on demand, though those should report "TEST SUCCESS".

> These ones are probably due to the same reason I wasn't able to use the email services with my gmail account, with gmail replying to me asking to use an app with more modern security standards. I will open a separate issue for this.

Not really. This tests for invalid credentials and it should report "TEST SUCCESS".

> About the integration tests there is no such file as rapp_run_tests.py anywhere in the 
${HOME}/rapp_platform/rapp-platform-catkin-ws
directory nor its subdirectories, so I did not manage to run them.

Usage of the rapp_testing_tools is well documented [here](https://github.com/rapp-project/rapp-platform/tree/devel/rapp_testing_tools)

lpenteri, 2016-05-10:
[test_output.txt](https://github.com/rapp-project/rapp-platform/files/256968/test_output.txt)

You can find attached my test output. There indeed are a few tests that failed nothing related to the weather nodes though.

When executing the integration tests
`rosrun rapp_testing_tools rapp_run_test.py`
this is the output I get:
[integration_tests_out.txt](https://github.com/rapp-project/rapp-platform/files/257031/integration_tests_out.txt)

Do these tests pass on Travis also with https enabled? Here where I have it enabled they fail and tls seems to be the reason.

klpanagi, 2016-05-10:
Thanks for the report @lpenteri . This seems to be a bug related to the rapp-platform-api and not the rapp-platform.

Those issues are no related to the issue title so i am closing this.
Please open a related issue in the [rapp-api](https://github.com/rapp-project/rapp-api) repository.

closed 2016-05-10 by klpanagi