rapp-project/rapp-platform #300: Update all READMEs, wiki pages etc

athallas, 2016-05-12:


etsardou, 2016-05-13:
## READMEs

- [ ] [Main README](https://github.com/rapp-project/rapp-platform/blob/master/README.md) - Manos
- [ ] Audio processing readme - Aris / Manos
- [ ] Cognitive exercize readme - Thanos
- [ ] Face detection readme - Manos
- [ ] Knowrob wrapper readme - Thanos
- [ ] Mysql wrapper readme - Thanos
- [ ] QR detection readme - Manos
- [ ] RAPP Scrips - Manos / Aris 
- [ ] Speech detection Google & Sphinx - Manos / Aris
- [ ] Rapp testing tools - Kwstas
- [ ] Text to speech espeak - Manos
- [ ] Web services  - Kwstas
- [ ] Caffe - Thanos
- [ ] Authentication manager - Aris
- [ ] Email - Aris
- [ ] Geolocator - Aris
- [ ] Hazard detection - Manos
- [ ] Human detection - Manos
- [ ] News explorer - Aris
- [ ] Path planning - Manos
- [ ] Utilities - Aris
- [ ] Weather reporter - Aris

## Wiki

- [ ] Architecture - Manos
- [ ] Multi threading issues - Kwstas / Aris
- [ ] HOP services - Kwstas
- [ ] Tutorials - General - Before the installation - Thanos
- [ ] Tutorials - After the installation - Thanos
- [ ] Ways to improve rapp-platform - Thanos / Kwstas
- [ ] Full tutorials - RAPP Platform - Thanos / Kwstas
- [ ] Full tutorials  - Robotic Applications (1-4) - Manos
- [ ] Full tutorials  - Robotic Applications (5) - Kwstas
- [ ] Full tutorials  - Cloud agents - Aris

akintsakis, 2016-05-14:
## Update Code Documentation
- [ ] Knowrob Wrapper - Thanos
- [ ] MySQL Wrapper - Thanos
- [ ] Caffe - Thanos
- [ ] Cognitive Exercise - Thanos

athallas, 2016-05-19:
This is obsolete:

https://github.com/rapp-project/rapp-platform/wiki/Create-and-authenticate-a-new-RAPP-user

klpanagi, 2016-05-21:
**Ways to improve rapp-platform**:

- [ ] How to create a HOP service for a ROS service?
- [ ] How to write the API for a HOP service?
- [ ] How to call the HOP service I created?
- [ ] How to implement an integration test?

closed 2016-11-08 by etsardou