rapp-project/rapp-platform #333: Web services having files as input crash when the file does not exist

etsardou, 2016-06-08:
For example in a ```setNoiseProfile``` call:

> response = self.ch.setNoiseProfile('vim.exe', 'nao_wav_1_ch')

I get:

```
Traceback (most recent call last):
  File "./denoise_profile_tests.py", line 45, in test_setNoiseProfile_noExistentFile
    response = self.ch.setNoiseProfile('vim.exe', 'nao_wav_1_ch')
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/RappPlatformApi.py", line 120, in setNoiseProfile
    response = self.svc_caller.call(msg)
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/Service/RappPlatformService.py", line 153, in call
    cloudResponseDic = self.__controller.run_job()
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/ServiceController/ServiceControllerSync.py", line 73, in run_job
    resp = self.__post_persistent(payload, files)
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/ServiceController/ServiceControllerSync.py", line 165, in __post_persistent
    return self.post_request(self.session_, data, files)
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/ServiceController/ServiceControllerSync.py", line 108, in post_request
    _files.append(f.make_tuple())
  File "/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-api/python/RappCloud/Objects/File.py", line 99, in make_tuple
    return (self.__postfield, (filename, open(self.__path)))
IOError: [Errno 2] No such file or directory: '/home/manos/rapp_platform/rapp-platform-catkin-ws/src/rapp-platform/rapp_testing_tools/scripts/default_tests/vim.exe'
```

Normally, an error should be returned to the ```error``` field of the response.

klpanagi, 2016-06-08:
This issue is related to the [python-platform-api](https://github.com/rapp-project/rapp-api/tree/devel/python)

So an **IOError Exception** is thrown if the file does not exist

etsardou, 2016-06-08:
Then we should internally add try-catch in order to return the error to the user.

closed 2016-07-01 by athallas