rapp-project/rapp-platform #337: Speech-Recognition-Google ROS Service does not return error on wrong language input

klpanagi, 2016-06-15:
Passing wrong language (`string`) does not return error. It instead uses default language, `'en'`

This was investigated on execution of the speech_recognition_google integration tests on [this](https://github.com/rapp-project/rapp-platform/blob/integration_tests_issue_285/rapp_testing_tools/scripts/default_tests/speech_detection_google_tests.py#L66) call, where `language='klklklkl'`

Parameters passed from client to the ROS Service are correct:

```bash
1 REQUEST.serv: POST *://anonymous@localhost:9001 /hop/speech_detection_google HTTP/1.1
1 INVOKE.svc (speech_detection_google . #|JsObject [__proto__: JsObject] [extensible: #t] [properties: ()] [cmap: JsConstructMap] [elements: vector]|)
{ filename: '/home/klpanagi/.hop/cache/server/9001/upload/recording_sentence1-KXswH5ISwIutyFma3L7vaYGuPY6F15..ogg',
  audio_type: 'nao_ogg',
  user: 'rapp',
  language: 'klklklkl' }
```

klpanagi, 2016-06-15:
Same for [this](https://github.com/rapp-project/rapp-platform/blob/integration_tests_issue_285/rapp_testing_tools/scripts/default_tests/speech_detection_google_tests.py#L68) call test, where `language=''`

klpanagi, 2016-06-29:
A `wrong language` error must be returned if input language value is invalid.

[CI fail](https://travis-ci.org/rapp-project/rapp-platform/jobs/141130334#L2493) for `language="klklklkl"`

athallas, 2016-06-30:
1. The commit message says "Error is returned when **no language** is specified" which is the reason this has not been merged in devel yet.
2. This will probably not be fixed since google returns no error when wrong language is specified. API does not provide any details and seems to be deprecated.
3. Cherry-picking commits can create duplications, so be careful with your branch structure.

commit 012ccd46 (2016-06-30): Error is return when no language is specified. (closes #337)
commit 394b1332 (2016-07-01): Error is return when no language is specified. (closes #337)
commit a1f915a2 (2016-07-01): Merge branch 'google_speech_detection_fix' into devel

* google_speech_detection_fix:
  Error is return when no language is specified. (closes #337)
closed 2016-07-01 by athallas