rapp-project/rapp-platform #340: Release v0.6

athallas, 2016-07-01:
- [ ] Create .ova file
- [ ] Deploy platform on server
- [ ] ROS News announcement

etsardou, 2016-07-01:
- [ ] @athallas  check https://github.com/rapp-project/rapp-platform/wiki/RAPP-Application-Authentication
- [ ] @akintsakis check https://github.com/rapp-project/rapp-platform/wiki/RAPP-Cognitive-Exercise for the new services
- [ ] @akintsakis check https://github.com/rapp-project/rapp-platform/wiki/RAPP-MySQL-wrapper
- [ ] Erroneous link in README of https://github.com/rapp-project/rapp-platform/tree/devel/rapp_text_to_speech_espeak

etsardou, 2016-07-12:
@klpanagi @akintsakis Status of the check list?

etsardou, 2016-07-12:
- [ ] Release rapp-platform
- [ ] Release rapp-api
- [ ] Release rapp-robot-api
- [ ] Release rapp-platform-scripts

klpanagi, 2016-07-12:
Need to updade testing-tools related documentation due to the latest changes;

etsardou, 2016-07-19:
@akintsakis Update [this](https://github.com/rapp-project/rapp-platform/wiki/RAPP-Cognitive-Exercise) with the latest services to proceed with the release.

closed 2016-11-18 by etsardou