rapp-project/rapp-platform #350: Protocol error with `email_send`

MariaRamos89, 2016-09-08:
Hello,

I have implemented the `email_send` C++ API in branch `cpp_dev` and the actual example isn't working.

It is returning a `500 Internal Server Error` HTTP header.
The gmail account is valid, I've tested it using curl with a **valid** gmail account, and it worked:

```
curl smtps://smtp.gmail.com:465 -v --mail-from "rappemailtest@gmail.com" --mail-rcpt "m.ramos@ortelio.co.uk" --ssl --user "rappemailtest@gmail.com:XXX" -T "test.cpp" -k --anyauth
```

However, the output from [examples/email_handling.cpp](https://github.com/rapp-project/rapp-api/blob/cpp_dev/cpp/examples/email_handling.cpp) is:

```
POST /hop/email_send HTTP/1.1
Host: rapp.ee.auth.gr:9001
Accept-Token: rapp_token
Content-Length: 261
Content-Type: multipart/form-data; boundary=SDJ8uuPl7CyFAx4N

--SDJ8uuPl7CyFAx4N
Content-Disposition: form-data; name="json"

{"body":"body mail", "email":"rappemailtest@gmail.com", "passwd":"XXX", "port":"465", "recipients":["m.ramos@ortelio.co.uk"], "server":"smtp.gmail.com", "subject":"test"}
--SDJ8uuPl7CyFAx4N--
```

Please advise.

klpanagi, 2016-09-14:
There exist a bug in the email_send web service implementation, having that service throwing an exception when no file is attached to the request body.
We are going to prepare a **bug-fix** for this

klpanagi, 2016-09-14:
Master fix patch at [fix_issue_350](https://github.com/rapp-project/rapp-platform/tree/fix_issue_350) branch.

Please checkout to that branch for testing with `email_send` service

etsardou, 2016-11-08:
@klpanagi if this is resolved push it to master

MariaRamos89, 2016-11-08:
For me, it's working :)
Thanks

commit d07d7fe3 (2016-11-08): Fixes #350 [ci skip]
closed 2016-11-08 by MariaRamos89