rapp-project/rapp-platform #353: [Web Services]: Allow for enabling verbose mode

klpanagi, 2016-09-25:


etsardou, 2016-11-08:
@maciek-slon what exactly did you mention about the verbose mode? What kind of additional information you wanted to be seen?

maciek-slon, 2016-11-09:
At the moment I can see something like this when calling service:

```
1 INVOKE.svc (object_detection_find_objects . #|JsObject [__proto__: JsObject] [extensible: #t] [properties: ()] [cmap: JsConstructMap] [elements: vector]|)
```

It would be nice to have possibility to see full request and response, or at least JSONs that are received and sent back

etsardou, 2016-11-09:
@klpanagi is this possible?

klpanagi, 2016-11-09:
Yeap, we will have to update **only** the [WebService HOP wrapper](https://github.com/rapp-project/rapp-platform/blob/master/rapp_web_services/src/webService/webService.js) and include a logger.
I will also add an option to redirect logs to either **stdout** or **log-files**.
Does not require too much effort, so i think an update will be ready by the end of the week

klpanagi, 2016-11-11:
Logging  improvements include:
- Added a proper logger that captures request and response descriptors. Includes timestamps and pretty-print.
- **Debug mode**:  Stream logs to stdout. Enable/Disable via config file [here](https://github.com/rapp-project/rapp-platform/blob/logging_web_svc/rapp_web_services/env.js#L62)
- **Log-Files**:  Stream logs to files. Enable/Disable usage of log-files via config file [here](https://github.com/rapp-project/rapp-platform/blob/logging_web_svc/rapp_web_services/env.js#L63)
- Lower HOP Verbosity.

Enabling both **Debug** and **Log-Files** will stream logs to both stdout and log-files

@maciek-slon Please chechout to that branch (`logging_web_svc`) and let us know if logging is sufficient in order to perform a merge to `master` branch

closed 2017-03-13 by klpanagi