rapp-project/rapp-platform #354: Error object_recognition

MariaRamos89, 2016-09-26:
Hi,

I'm running an example of `object_recognition` and the answer from the platform is this:

```
Error JSON: service [/rapp/rapp_caffe_wrapper/image_classification] responded with an error: error processing request: broken PNG file (chunk 'c4{0')
Found hand blower, blow dryer, blow drier, hair dryer, hair drier
Error JSON: service [/rapp/rapp_caffe_wrapper/image_classification] responded with an error: error processing request: broken PNG file (chunk '5@d\xb7')
Found hand blower, blow dryer, blow drier, hair dryer, hair drier
Found hand blower, blow dryer, blow drier, hair dryer, hair drier
Error JSON: service [/rapp/rapp_caffe_wrapper/image_classification] responded with an error: error processing request: broken PNG file (chunk 'U\xd8\xe6g')
```

I'm sending images from my usb camera, and I don't know why sometimes works and sometimes not. 
In any case, the algorithm doesn't work properly because there is no hair dryer in the image.

Thanks in advance

etsardou, 2016-09-26:
Hey, I have no idea why the service responds with error AND the results.. Maybe @klpanagi can help.

Concerning the correctness of the algorithm, this node wraps the Caffe framework with its default trained model, so it is highly unlikely that it will produce the correct results (as it does not contain all possible object classes) and unless the object is clearly visible and dominant in the frame.

klpanagi, 2016-09-26:
Dear Maria,

Do you specify the `filename` parameter in post form? If yes and it is identical for every request, then this is likely to be the issue. If you provide that parameter you force the Web Server to use that value for storing the file. This can lead to similar errors because:

- The file gets overwritten
- The Object-Detection Node tries to read from a file while the Web Server writes, **"from the next"** request

MariaRamos89, 2016-09-29:
Maybe it's the second one.
The program I was running take a picture from a webcam and send it to do object recognition every 2-5 seconds (depending of the resolution). It's too much time so I don't know.

klpanagi, 2016-09-29:
**The question is**: Does the client-api apply the same value for the `filename` parameter of the post form, for each request made to the Platform?

MariaRamos89, 2016-09-29:
No, the filename is a random string every time, it uses the same mt19937 PRNG as the one used by the boundary for multipart forms.