rapp-project/rapp-platform #359: The new wut services related to object detection ask for a parameter 'user'

lpenteri, 2016-11-22:
Wasn't this parameter removed from the requests of all services and it was supposed to be retrieved internally in the platform after a login call and the authentication of every request would be done through the token?

klpanagi, 2016-11-22:
Hello @lpenteri ,

what do you mean "asks" for parameter `user`? From where did you retrieve this information?
Look [here](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/object_detection_clear_models/svc.js#L48). The username is retrieved from the authentication server after validating the `token`.

Though, @maciek-slon, the `user` parameter has not been removed from the `client_req` object of object_detection related services:

- [object_detection_clear_models](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/object_detection_clear_models/iface_obj.js#L33)
- [object_detection_find_objects](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/object_detection_find_objects/iface_obj.js#L38)
- [object_detection_learn_object](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/object_detection_learn_object/iface_obj.js#L35)
- [object_detection_load_models](https://github.com/rapp-project/rapp-platform/blob/devel/rapp_web_services/services/object_detection_load_models/iface_obj.js#L34)

Though this is not a problem as the `user` parameter is not used...

@maciek-slon  we also have to update the WebServices list and documentation [here](https://github.com/rapp-project/rapp-platform/tree/devel/rapp_web_services/services#rapp-platform-web-services) to include the **object_detection** related services

maciek-slon, 2016-11-22:
@klpanagi can I push directly to devel after README change? Or should we do a whole round of branch/pull request/merge? ;-)

lpenteri, 2016-11-22:
Hi @klpanagi ,

Yes that's why I thought I have to send it, because I saw it in the client_req objects.
Ok then it's not a problem I will not send it. :)

klpanagi, 2016-11-22:
@maciek-slon create a new branch, having `devel` as parent and push changes there :)

maciek-slon, 2016-11-22:
OK. See #360

closed 2016-11-22 by klpanagi