rapp-project/rapp-platform #361: mutli-robot support

ShawnSchaerer, 2017-03-10:
Question regarding rapp platform.
Are there any examples of multi-robot support for rapps?

etsardou, 2017-03-15:
Hi Shawn! Can you tell us a bit more on how you imagine to use it in order to help you?

RAPP Platform is deployed in the cloud, so any robot (or device) can invoke the services, provided the correct credentials.

If you ask whether a robotic application can manage two robots at the same time, it can by instantiating two instances of the RAPP Robot API (https://github.com/rapp-project/rapp-robots-api/tree/master/python), declaring the two different IPs.

ShawnSchaerer, 2017-03-15:
Hi,
I was thinking about using it as a master that would talk to more than one robot at a time.  
I will take a look at the API